mpromonet/webrtc-streamer #714: cmake this project have red error in freebsd 14.1

runsys, 2024-08-16:
root@BPC:/media/Generic_MassStorageClass_000000001539_s1/work/code/qt/webrtc-streamer-master # cmake .
-- The C compiler identification is Clang 18.1.5
-- The CXX compiler identification is Clang 18.1.5
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /media/Generic_MassStorageClass_000000001539_s1/work/code/qt/webrtc-streamer-master/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=Clang
-- Found Git: /usr/local/bin/git (found version "2.45.2")
fatal: not a git repository (or any parent up to mount point /media)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /media)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /media)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: cannot change to '/media/Generic_MassStorageClass_000000001539_s1/work/code/qt/webrtc-streamer-master/../webrtc/src': No such file or directory
fatal: not a git repository (or any parent up to mount point /media)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = TRUE
PulseAudio_FOUND = 1
CMake Error at CMakeLists.txt:91 (file):
  file failed to open for reading (No such file or directory):

    /media/Generic_MassStorageClass_000000001539_s1/work/code/qt/webrtc-streamer-master/../webrtc/src/modules/congestion_controller/goog_cc/loss_based_bwe_v2.cc

GTK3_FOUND = 1
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
RTMP_FOUND =  RTMP_INCLUDE_DIRS= RTMP_LIBRARY_DIRS= RTMP_LIBRARIES=
CMake Error at CMakeLists.txt:254 (add_subdirectory):
  add_subdirectory given source "prometheus-cpp/core" which is not an
  existing directory.

CMake Error at CMakeLists.txt:260 (add_subdirectory):
  The source directory

    /media/Generic_MassStorageClass_000000001539_s1/work/code/qt/webrtc-streamer-master/live555helper

  does not contain a CMakeLists.txt file.

-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC - Success

mpromonet, 2024-08-16:
Hi @runsys 
It seems the error message says you did not installed webrtc sdk
Best Regards 
Michel

runsys, 2024-08-16:
yes,now ok

closed 2024-08-16 by runsys