mpromonet/webrtc-streamer #718: Screen flickering or distorted display

3-7k, 2024-08-20:
When I use the -o parameter, the screen often flickers and turns green, but this doesn't happen when I don't use the -o parameter. How can I solve this issue?

[image]

[image]

mpromonet, 2024-09-07:
Hi,

It might be SPS/PPS parsing ?

Best Regards,
Michel.

BlockCnFuture, 2024-09-07:
> Hi, 你好，
> 
> It might be SPS/PPS parsing ?可能是SPS/PPS解析？
> 
> Best Regards, 此致， Michel. 米歇尔.

I have the same problem, is there a solution? Thanks

mpromonet, 2024-09-07:
Hi @BlockCnFuture 

You can look of buffer in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h#L107 and look how are processed SPS/PPS and buffer send in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h#L157C21-L157C28

Best Regards,
Michel.

3-7k, 2024-09-14:
@mpromonet thanks for your help

closed 2024-09-14 by 3-7k

flowersBloom826, 2025-04-25:
@3-7k Boss, then how to solve it?