mpromonet/webrtc-streamer #723: ./webrtc-streamer: error while loading shared libraries: libSM.so.6: cannot open shared object file: No such file or directory

Songgexnl, 2024-09-25:
[image]
下载的这个，然后在 x86_64 x86_64 x86_64 GNU/Linux 环境中运行报错./webrtc-streamer: error while loading shared libraries: libSM.so.6: cannot open shared object file: No such file or directory
；并且我已安装了
[image]
执行 ./webrtc-streamer
还是报错 error while loading shared libraries: libSM.so.6: cannot open shared object file: No such file or directory

mpromonet, 2024-09-26:
Hi @Songgexnl 
Binaries for Linux are built on Ubuntu 24.04.
Best Regards 
Michel

Songgexnl, 2024-09-26:
> Hi @Songgexnl Binaries for Linux are built on Ubuntu 24.04. Best Regards Michel

Do you have this installation package for x86_64 x86_64 GNU/Linux environment

mpromonet, 2024-09-28:
Hi @Songgexnl 

Did you look to https://amalgjose.com/2020/03/14/importerror-libsm-so-6-cannot-open-shared-object-file-no-such-file-or-directory/ ?

Best Regards,
Michel.

closed 2024-11-11 by mpromonet