mpromonet/webrtc-streamer #729: can't run on servers

aws0me, 2024-10-31:
i run it in docker on my mac,it works.
i run it in docker on a server without internet,it doesn't work.
also,how to improve the log level,

here is the log:

[root@i-O2onpOhgE ~]# docker logs -f 53455dcafccb8b6e8bb9bd23fc0ba5120876cf2ee851f03b1f5a6e43aa676483
Version:v0.8.7-15-gdadf698/Linux-x86_64 civetweb@v1.16 webrtc@9912723-dirty live555helper@63e2d96
{
        "urls" : 
        {
                "tianyan" : 
                {
                        "video" : "rtsp://10.230.1.215:8554/live/JJ-JH-06"
                }
        }
}Logger level:4
HTTP Listen at 0.0.0.0:8000
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Start playing sink for "video/H264" subsession
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Close session: video/H264
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Start playing sink for "video/H264" subsession
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Close session: video/H264
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Start playing sink for "video/H264" subsession
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Close session: video/H264
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Start playing sink for "video/H264" subsession
Requested URL : rtsp://10.230.1.215:8554/live/JJ-JH-06
Close session: video/H264

image version:mpromonet/webrtc-streamer:master-amd64

corrosive4354, 2024-11-01:
The current Docker version may have bugs related to RTSP compatibility.

I think you can try the [servbay](https://servbay.com) tool, which is more professional on Mac and supports more than docker.

mpromonet, 2024-11-01:
Hi @corrosive4354,

Which kinds of bugs, probably the backend is just not getting data from RTSP.

Best regards,
Michel.

LiYuTong888, 2025-04-17:
It's like this, I seem to have encountered the same problem. I obtained an RTSP stream from a webcam and handed it over to a WebRTC streamer to play. At first, the image was normal, although there were also error messages: “D:\work\webrtc-streamer-v0.8.10-Debug\bin>webrtc-streamer.exe -H 8001 -v 3 --force_tcp -o
Version:v0.8.10-dirty/Windows-AMD64  civetweb@v1.16   webrtc@4030180fc5-dirty   live555helper@bd5f554
{
"urls" :
{
"--force_tcp" :
{
"video" : "--force_tcp"
},
"-o" :
{
"video" : "-o"
},
"3" :
{
"video" : "3"
}
}
}Logger level:3
[000:000][109916] (audio_device_generic.cc:28): virtual webrtc::AudioDeviceGeneric::BuiltInAGCIsAvailable: Not supported on this platform
[000:000][109916] (audio_device_generic.cc:38): virtual webrtc::AudioDeviceGeneric::BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 8001
[005:761][109648] (PeerConnectionManager.h:274): virtual PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded peerid:0.16626948521689489
[005:763][109916] (CapturerFactory.h:301): audiourl: idx_audioDevice:-1/1
[005:764][110168] (PeerConnectionManager.cpp:1335): Cannot create capturer audio:
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1
Start playing sink for "video/H264" subsession
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1
[005:830][109916] (video_send_stream_impl.cc:253): ERROR: Initial encoder max bitrate = -1 which is <= 0!
[005:850][109648] (PeerConnectionManager.h:183): virtual PeerConnectionManager::DataChannelObserver::OnStateChange channel:ServerDataChannel state:open
[005:850][109648] (PeerConnectionManager.h:270): virtual PeerConnectionManager::PeerConnectionObserver::OnDataChannel
[005:850][109648] (PeerConnectionManager.h:190): virtual PeerConnectionManager::DataChannelObserver::OnMessage channel:ServerDataChannel msg:remote channel openned
[005:851][109648] (PeerConnectionManager.h:183): virtual PeerConnectionManager::DataChannelObserver::OnStateChange channel:ClientDataChannel state:open
[005:851][109648] (PeerConnectionManager.h:190): virtual PeerConnectionManager::DataChannelObserver::OnMessage channel:ClientDataChannel msg:local channel openned
[045:591][105080] (stun_port.cc:102): Binding request timed out from 192.168.4.x:50143 (WLAN)”

After a period of time, the screen froze and then reported this message. 

How can I solve it :“Close session: video/H264
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1
Start playing sink for "video/H264" subsession
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1
Close session: video/H264
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1
Start playing sink for "video/H264" subsession
Requested URL :  rtsp://admin:123456 @192.168.4.85:554/stream1”

closed 2026-10-31 by mpromonet