mpromonet/webrtc-streamer #731: When playing RTSP monitoring, the screen occasionally freezes, and the reason for the original monitoring stream has been ruled out.

bijianhua123, 2024-10-31:


mpromonet, 2024-11-01:
Hi @bijianhua123 

Thanks for your issue, but if you like some answer you need to clarify.... what is RTSP monitoring...

Best Regards,
Michel.

bijianhua123, 2024-11-07:
Hikvision surveillance video stream，It is RTSP type

mpromonet, 2024-11-08:
Hi @bijianhua123 

Wireshark or tcpdump may helps you.

Best Regards 
Michel

closed 2024-11-08 by mpromonet

yangjieshao, 2024-11-08:
> surveillance

直连设备还是通过isc平台？
通过isc平台好像是有问题，isc返回的rtsp地址是加密后的，是有有效期的，理论上只要连上了就可以一直放
但是不知道为什么 用webrtc-streamer，地址过期后就会断开

bijianhua123, 2024-11-12:
isc platform。Have you also encountered this problem? Is there any solution? @yangjieshao

yangjieshao, 2024-11-12:
> isc platform。Have you also encountered this problem? Is there any solution? @yangjieshao

https://github.com/mpromonet/webrtc-streamer/issues/654

最终是和业主协商，直连设备才解决的
It was ultimately resolved through negotiation with the property owner and direct connection to the equipment

zhuya1996, 2025-01-08:
需要自己编译的webrtc 改里面代码 就解决了 我也遇到了

a9010028, 2025-03-24:
@zhuya1996 請問您是修改哪邊東西呢 是否能分享一下,感謝(May I ask which part you modified? Would you mind sharing it with me? Thank you!)

zhuya1996, 2025-03-24:
> [@zhuya1996](https://github.com/zhuya1996) 請問您是修改哪邊東西呢 是否能分享一下,感謝(May I ask which part you modified? Would you mind sharing it with me? Thank you!)

Need to add my QQ 2308193190 Not free

hansongLove, 2025-07-22:
> isc平台。你也遇到过这个问题吗？有什么解决办法吗？[@yangjieshao](https://github.com/yangjieshao)

Have you solved this problem? Can you help me with it? Thank you.

zhuya1996, 2025-07-22:
> @zhuya1996请问您是修改哪边东西呢 是否能分享一下，谢谢（请问您修改了哪一部分？您介意与我分享吗？谢谢！）

你是中国人吧，是不是在csdn上已经问过我了