mpromonet/webrtc-streamer #741: ICE LocalNetwork

aad-0, 2024-12-05:
It seems like it has trouble to request stream with no internet connection. ICE server causes problem. Can you help please?

aad-0, 2024-12-05:
Okay, I dont think Ice is problem, but output from webrtcstreamer.js is connect error: InvalidStateError: Can't create RTCPeerConnections when the network is down

mpromonet, 2024-12-06:
Hi,
I guess you need a stun server when you cannot reach internet. 
Using argument -S may help.
Best Regards 
Michel

aad-0, 2024-12-09:
Logger level:1
[000:000][145810] (field_trial.cc:164): Setting field trial string:WebRTC-FrameDropper/Disabled/
[000:000][145811] (audio_device_buffer.cc:66): AudioDeviceBuffer::ctor
[000:000][145811] (audio_device_impl.cc:146): current platform is Linux
[000:000][145811] (audio_device_impl.cc:168): CreatePlatformSpecificObjects
[000:000][145811] (audio_device_impl.cc:906): PlatformAudioLayer
[000:000][145811] (audio_device_impl.cc:226): PulseAudio support is enabled.
[000:000][145811] (audio_device_impl.cc:231): Linux PulseAudio APIs will be utilized
[000:000][145811] (audio_device_impl.cc:273): AttachAudioBuffer
[000:000][145811] (audio_device_buffer.cc:197): SetRecordingSampleRate(0)
[000:000][145811] (audio_device_buffer.cc:203): SetPlayoutSampleRate(0)
[000:000][145811] (audio_device_buffer.cc:217): SetRecordingChannels(0)
[000:000][145811] (audio_device_buffer.cc:223): SetPlayoutChannels(0)
[000:000][145811] (audio_device_impl.cc:293): Init
[000:004][145810] (input_volume_stats_reporter.cc:98): Will not log any `WebRTC.Audio.Apm.AppliedInputVolume.*` histogram stats.
[000:004][145810] (input_volume_stats_reporter.cc:98): Will not log any `WebRTC.Audio.Apm.RecommendedInputVolume.*` histogram stats.
[000:004][145810] (audio_processing_impl.cc:461): Injected APM submodules:
Echo control factory: 0
Echo detector: 0
Capture analyzer: 0
Capture post processor: 0
Render pre processor: 0
[000:004][145810] (audio_processing_impl.cc:474): AudioProcessing: AudioProcessing::Config{ pipeline: { maximum_internal_processing_rate: 48000, multi_channel_render: 0, multi_channel_capture: 0 }, pre_amplifier: { enabled: 0, fixed_gain_factor: 1 },capture_level_adjustment: { enabled: 0, pre_gain_factor: 1, post_gain_factor: 1, analog_mic_gain_emulation: { enabled: 0, initial_level: 255 }}, high_pass_filter: { enabled: 0 }, echo_canceller: { enabled: 0, mobile_mode: 0, enforce_high_pass_filtering: 1 }, noise_suppression: { enabled: 0, level: Moderate }, transient_suppression: { enabled: 0 }, gain_controller1: { enabled: 0, mode: AdaptiveAnalog, target_level_dbfs: 3, compression_gain_db: 9, enable_limiter: 1, analog_gain_controller { enabled: 1, startup_min_volume: 0, clipped_level_min: 70, enable_digital_adaptive: 1, clipped_level_step: 15, clipped_ratio_threshold: 0.1, clipped_wait_frames: 300, clipping_predictor:  { enabled: 0, mode: 0, window_length: 5, reference_window_length: 5, reference_window_delay: 5, clipping_threshold: -1, crest_factor_margin: 3, use_predicted_step: 1 }}}, gain_controller2: { enabled: 0, fixed_digital: { gain_db: 0 }, adaptive_digital: { enabled: 0, headroom_db: 5, max_gain_db: 50, initial_gain_db: 15, max_gain_change_db_per_second: 6, max_output_noise_level_dbfs: -50 }, input_volume_control : { enabled 0}}
[000:004][145816] (webrtc_voice_engine.cc:464): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:004][145811] (webrtc_voice_engine.cc:486): WebRtcVoiceEngine::Init
[000:004][145811] (audio_device_impl.cc:293): Init
[000:004][145811] (audio_device_impl.cc:637): SetPlayoutDevice(0)
[000:004][145811] (audio_device_impl.cc:326): InitSpeaker
[000:005][145811] (audio_device_impl.cc:541): StereoPlayoutIsAvailable
[000:005][145811] (audio_device_impl.cc:548): output: 1
[000:005][145811] (audio_device_impl.cc:553): SetStereoPlayout(1)
[000:005][145811] (audio_device_buffer.cc:223): SetPlayoutChannels(2)
[000:005][145811] (audio_device_impl.cc:699): SetRecordingDevice(0)
[000:005][145811] (audio_device_impl.cc:332): InitMicrophone
[000:005][145811] (audio_device_impl.cc:495): StereoRecordingIsAvailable
[000:005][145811] (audio_device_impl.cc:502): output: 1
[000:005][145811] (audio_device_impl.cc:507): SetStereoRecording(1)
[000:005][145811] (audio_device_buffer.cc:217): SetRecordingChannels(2)
[000:005][145811] (audio_device_impl.cc:812): RegisterAudioCallback
[000:005][145811] (webrtc_voice_engine.cc:588): WebRtcVoiceEngine::ApplyOptions: AudioOptions {aec: 1, agc: 1, ns: 1, hf: 1, swap: 0, audio_jitter_buffer_max_packets: 200, audio_jitter_buffer_fast_accelerate: 0, audio_jitter_buffer_min_delay_ms: 0, }
[000:005][145811] (audio_device_impl.cc:828): BuiltInAECIsAvailable
[000:005][145811] (audio_device_generic.cc:18): BuiltInAECIsAvailable: Not supported on this platform
[000:005][145811] (audio_device_impl.cc:831): output: 0
[000:005][145811] (audio_device_impl.cc:844): BuiltInAGCIsAvailable
[000:005][145811] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:005][145811] (audio_device_impl.cc:847): output: 0
[000:005][145811] (audio_device_impl.cc:860): BuiltInNSIsAvailable
[000:005][145811] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
[000:005][145811] (audio_device_impl.cc:863): output: 0
[000:005][145811] (audio_processing_impl.cc:673): AudioProcessing::ApplyConfig: AudioProcessing::Config{ pipeline: { maximum_internal_processing_rate: 48000, multi_channel_render: 0, multi_channel_capture: 0 }, pre_amplifier: { enabled: 0, fixed_gain_factor: 1 },capture_level_adjustment: { enabled: 0, pre_gain_factor: 1, post_gain_factor: 1, analog_mic_gain_emulation: { enabled: 0, initial_level: 255 }}, high_pass_filter: { enabled: 1 }, echo_canceller: { enabled: 1, mobile_mode: 0, enforce_high_pass_filtering: 1 }, noise_suppression: { enabled: 1, level: High }, transient_suppression: { enabled: 0 }, gain_controller1: { enabled: 1, mode: AdaptiveAnalog, target_level_dbfs: 3, compression_gain_db: 9, enable_limiter: 1, analog_gain_contr [... cut ...]
[000:006][145811] (agc_manager_direct.cc:481): [agc] analog controller enabled: yes
[000:006][145811] (agc_manager_direct.cc:484): [agc] Min mic level: 12 (overridden: no)
HTTP Listen at 127.0.0.1:9999
STUN Listening at 0.0.0.0:3478

[008:550][145819] (HttpServerRequestHandler.cpp:61): uri:/api/getIceServers
[008:551][145819] (PeerConnectionManager.cpp:560): ICE URL:stun:127.0.0.1:3478
[016:238][145823] (HttpServerRequestHandler.cpp:61): uri:/api/getIceServers
[016:239][145823] (PeerConnectionManager.cpp:560): ICE URL:stun:127.0.0.1:3478

[image]

aad-0, 2024-12-09:
Stun Server seems up, yet cannot create peer connection

closed 2025-05-17 by mpromonet