mpromonet/webrtc-streamer #744: chrome is supported h265

liusuyi2021, 2024-12-11:
Supported browsers
The server can ingest and broadcast with WebRTC a wide variety of video and audio codecs (that are listed at the beginning of the README), but not all browsers can publish and read all codecs due to internal limitations that cannot be overcome by this or any other server.

In particular, reading and publishing H265 tracks with WebRTC was not possible until some time ago due to the lack of browser support. The situation recently improved and can be described as following:

Safari on iOS and macOS fully supports publishing and reading H265 tracks

Chrome on Windows supports publishing and reading H265 tracks when a GPU is present and when the browser is launched with the following flags:
chrome.exe --enable-features=PlatformHEVCEncoderSupport,WebRtcAllowH265Receive,WebRtcAllowH265Send --force-fieldtrials=WebRTC-Video-H26xPacketBuffer/Enabled

Mediamtx already supports the H265 WebRTC protocol, can WebRTC streamer also support it

mpromonet, 2024-12-11:
Hi @liusuyi2021 

It might work using -o to forward encoded stream and -W to set trial fields.

Best Regards 
Michel

osenbal, 2025-01-04:
do you know how to play h265 video in chrome?

osenbal, 2025-01-04:
Hi @mpromonet 
 
I'm trying to stream RTSP H265 Holowith CCTV in chrome browser with -o command. And i have check H265 codec in google chorme windows. But the stream wouldn't appear. 

[image]

if i missing something please let me know

mpromonet, 2025-01-04:
Hi @osenbal 

The support of h265 seems closed...
You could increase log level and check webrtc-internal to understand what fails.

Best Regards
Michel

osenbal, 2025-01-04:
@mpromonet 

So the problem is in browser (chrome) ?

this is the log verbose : 

[image]

mpromonet, 2025-01-04:
Hi @osenbal 

problems may be in each layer, it may be in webrtc-streamer, in webrtc sdk, in javascript side, in browser....
about://webrtc-internals may helps too.

Best Regards,
Michel.

osenbal, 2025-01-04:
@mpromonet so what should i do ? 

I stuck in here, i have try with mediamtx and its working but they  not provide components like webrtc-streamer

mpromonet, 2025-01-04:
Hi @osenbal 

what means stuck... 

mediamtx seems to support WHEP, you might look to https://github.com/Eyevinn/whep-video-component, that is also compatible with webrtc-streamer.

Best Regards,
Michel.

osenbal, 2025-01-04:
I tried to troubleshoot the issue with streaming H.265 RTSP with webrtc-streamer but got stuck.

I also attempted to use WHEP in webrtc-streamer to stream H.265 RTSP, but it didn’t work either.

mpromonet, 2025-01-04:
Hi @osenbal 

In commit https://github.com/mpromonet/webrtc-streamer/commit/213f22171b8cba634b58e8a6307bcb2b42acba00 I added a check that format from backend is compatible with webrtc format.

Best Regards,
Michel.

osenbal, 2025-01-04:
Have you tested with rtsp h265 with webrtc-streamer

mpromonet, 2025-01-04:
Hi @osenbal 

using https://github.com/mpromonet/go-whep-play h265 frames from rtsp server are received through webrtc.

Best Regards,
Michel.

osenbal, 2025-01-05:
Hi @mpromonet 
I think i got the problem from log on new version 0.8.9 

I noticed an issue in the Encode function related to the video format validation. Specifically, the error log:
format name must be {m_format.name} not {format.name}

[image]

maybe this is cause rtsp h265 cannot playing in thebrowser ?

mpromonet, 2025-01-05:
Hi @osenbal 

This error should occur once, to exclude h264 as a possible format. If it occurs many times, that probably means the offer send by web browser doesnot contains h265.
You should see this in /api/call.

Best Regards
Michel

mpromonet, 2025-01-19:
Hi,

commit 0c5f098c36885240fd006ece20495c33783e0608 add a test file for H265 codec.

Best Regards,
Michel.

ben-xD, 2025-03-07:
> Safari on iOS and macOS fully supports publishing and reading H265 tracks

@liusuyi2021, that can't be right? I tested the latest Chrome (`Version 134.0.6998.45 (Official Build) (arm64)`) today, and it logs:
```
[77331:64099:0307/154339.326058:ERROR:webrtc_video_engine.cc(1099)] No video codecs in common.
```

I was sending H265 over WebRTC (my custom server and client - not webrtc-streamer)

Therefore, it seems like even on macOS, you need:
```
--enable-features=WebRtcAllowH265Receive
```

ben-xD, 2025-05-17:
Good news! I just noticed today, these flags are not needed on Chrome for macOS or Windows - it works out of the box (I didn't test linux). Quickest way for you to test is looking at https://vdo.ninja/h265

lipengsen0812, 2025-05-20:
@ben-xD Hello, does Linux support H265 browser?

ben-xD, 2025-05-20:
Nope, it's not as of `136.0.7103.113`

mpromonet, 2026-10-13:
Hi,
I am closing this as it is working, latest with -o option on chrome 155 is displaying h265,
Best Regards,
Michel.

closed 2026-10-13 by mpromonet