mpromonet/webrtc-streamer #747: RTSP streams that are connected for more than 10 seconds cannot be played and will be pulled again

1910054472, 2024-12-31:
RTSP streams that are connected for more than 10 seconds cannot be played and will continue to be retried. How do I set the 10-second timeout period
[image]

mpromonet, 2024-12-31:
Hi @1910054472 
You can add timeout=60 in the webrtc url.
Maybe you should increase the log level to understand why the connection is closed (server side, no data received, missing rtcp, ....)
Best Regards,
Michel.

1910054472, 2025-01-02:
Hi @mpromonet 
This rtsp url can be played normally using VLC, but the buffer time is 15 seconds when it is opened, and the connection will be reconnected after 10 seconds when WebrTC-Streameris used.
add timeout=60 in the webrtc url，I haven't found where I can set it. Could you give me a hint
Best Regards,
fans

mpromonet, 2025-01-03:
Hi @1910054472 

The timeout could be passed as argument options in /api/call api. It depends how you open the connection...using whep...
An educated guess may be your rtsp server does not implement rtcp, and vlc implement a kind of keepalive, calling periodically rtsp getparameter... Hypothesis to verify ?

Best Regards
Michel

closed 2025-01-03 by 1910054472

liutianye, 2025-04-17:
Hey there! I’m running into the exact same problem that was mentioned in this thread. Just wanted to check if this issue has been resolved yet and, if so, how? Would you mind sharing what worked for you? Any help or advice would be super appreciated! Thanks a bunch!

mpromonet, 2025-04-17:
Hi @liutianye 

I guess rtsp server does not implemented rtcp ? One workaround could be to implement a cyclic getparameter.

Best Regards 
Michel

LiYuTong888, 2025-04-18:
> RTSP streams that are connected for more than 10 seconds cannot be played and will continue to be retried. How do I set the 10-second timeout period [image]

Hi，大佬，我也遇到了同样的问题，我看您得到解答后关闭了问题，您解决了吗

mpromonet, 2025-04-21:
Hi @liutianye 

This commit https://github.com/mpromonet/webrtc-streamer/commit/3fe5f802431905453af246d678825b640051194f should implement a keepalive in case RTCP is not working correctly. Of course the RTSP server need to support GET_PARAMETER keep alive.

Best Regards,
MicheL

liutianye, 2025-04-22:
> Hi [@liutianye](https://github.com/liutianye)
> 
> This commit [3fe5f80](https://github.com/mpromonet/webrtc-streamer/commit/3fe5f802431905453af246d678825b640051194f) should implement a keepalive in case RTCP is not working correctly. Of course the RTSP server need to support GET_PARAMETER keep alive.
> 
> Best Regards, MicheL

Wow, you're absolutely amazing—thank you so much! The current version is v0.8.10, and once your commit makes it into the new version (0.8.11) or there's an updated package available, I'll be sure to test it right away to see if the disconnection issue is resolved.

1910054472, 2025-04-22:
> > RTSP streams that are connected for more than 10 seconds cannot be played and will continue to be retried. How do I set the 10-second timeout period [image]
> 
> [@1910054472](https://github.com/1910054472) Hi， Boss, I also encountered the same problem. I saw that you closed the problem after receiving the answer. Have you solved it

我用WVP代理了一层

1910054472, 2025-04-22:
> [@1910054472](https://github.com/1910054472) Hi， Boss, I also encountered the same problem. I saw that you closed the problem after receiving the answer. Have you solved it

@LiYuTong888 @liutianye 没解决    我用WVP代理了一层

mpromonet, 2025-04-26:
Hi @liutianye 

A new release is available https://github.com/mpromonet/webrtc-streamer/releases/tag/v0.8.11 (except for macos)

Best Regards,
Michel.