mpromonet/webrtc-streamer #748: link bug

CalebShiffe, 2025-01-04:
my system is ubbuntu 22.04

i download webrtc  
fetch --no-history --nohooks webrtc
gn gen out/Release
ninja -C out/Release

/workspace/webrtc-streamer/build$ sudo cmake ..
/workspace/webrtc-streamer/build$ sudo make

[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/main.cpp.o
In file included from /home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/ssl_stream_adapter.h:29,
                 from /home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/ssl_adapter.h:21,
                 from /home/dltc/workspace/webrtc-streamer/src/main.cpp:15:
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h: In destructor ‘virtual rtc::StreamInterface::~StreamInterface()’:
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:57:30: warning: ‘rtc::StreamInterface::SignalEvent’ is deprecated: Use SetEventCallback instead [-Wdeprecated-declarations]
   57 |   virtual ~StreamInterface() {}
      |                              ^
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:105:48: note: declared here
  105 |   sigslot::signal3<StreamInterface*, int, int> SignalEvent
      |                                                ^~~~~~~~~~~
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:57:30: warning: ‘rtc::StreamInterface::SignalEvent’ is deprecated: Use SetEventCallback instead [-Wdeprecated-declarations]
   57 |   virtual ~StreamInterface() {}
      |                              ^
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:105:48: note: declared here
  105 |   sigslot::signal3<StreamInterface*, int, int> SignalEvent
      |                                                ^~~~~~~~~~~
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h: In member function ‘void rtc::StreamInterface::FireEvent(int, int)’:
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:138:5: warning: ‘rtc::StreamInterface::SignalEvent’ is deprecated: Use SetEventCallback instead [-Wdeprecated-declarations]
  138 |     SignalEvent(this, stream_events, err);
      |     ^~~~~~~~~~~
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:105:48: note: declared here
  105 |   sigslot::signal3<StreamInterface*, int, int> SignalEvent
      |                                                ^~~~~~~~~~~
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:138:5: warning: ‘rtc::StreamInterface::SignalEvent’ is deprecated: Use SetEventCallback instead [-Wdeprecated-declarations]
  138 |     SignalEvent(this, stream_events, err);
      |     ^~~~~~~~~~~
/home/dltc/workspace/webrtc-streamer/../webrtc/src/rtc_base/stream.h:105:48: note: declared here
  105 |   sigslot::signal3<StreamInterface*, int, int> SignalEvent
      |                                                ^~~~~~~~~~~
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtpvideocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
[100%] Linking CXX executable webrtc-streamer
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/rtc_base/rtc_json/json.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_reader.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_value.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_writer.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/p2p/p2p_server_utils/stun_server.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: /home/dltc/workspace/webrtc/src/out/Release/obj/p2p/p2p_server_utils/turn_server.o: unknown type [0x40000014] section `.crel.text'
/usr/bin/ld: ../../webrtc/src/out/Release/obj/api/video_codecs/libbuiltin_video_encoder_factory.a: error adding symbols: archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:258: webrtc-streamer] Error 1
make[1]: *** [CMakeFiles/Makefile2:124: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

 - Browser [firefox]

CalebShiffe, 2025-01-04:
@mpromonet  help me

CalebShiffe, 2025-01-04:
dltc@dltc-GeekPro-17IRR:~/workspace/webrtc/src$ gn gen out/Release --args='is_debug=false  target_os="linux" target_cpu="x64"'
Generating compile_commands took 71ms
Done. Made 2273 targets from 360 files in 1347ms
dltc@dltc-GeekPro-17IRR:~/workspace/webrtc/src$ ninja -C out/Release
ninja: Entering directory `out/Release'
[6388/6388] LINK ./video_engine_tests

CalebShiffe, 2025-01-04:
[image]

CalebShiffe, 2025-01-04:
[image]

mpromonet, 2025-01-04:
Hi @CalebShiffe 

Please look to readme file.
You built webrtc SDK with default options, then it use custom cxx lib. 
You may try to delete webrtc/out/Release and let cmake to build it.

Best Regards 
Michel

CalebShiffe, 2025-01-04:
Thank you, you are truly a sage. I discovered that this was a conflict between two compilers, and then I modified:

bash

gn gen out/Release-gcc --args='target_os="linux" target_cpu="x64" is_debug=false is_component_build=false use_sysroot=false is_clang=false use_lld=false treat_warnings_as_errors=false rtc_include_tests=false rtc_build_examples=false use_custom_libcxx=false use_rtti=true'
ninja -C out/Release-gcc

After compilation, I directly copied out/Release-gcc to out/Release. Then, when compiling webrtc-streamer again, the following issues were reported:sudo cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/dltc/workspace/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
WEBRTCVERSION=b00c469cad3f8c926fcf81ded90b90b6e1e62b9c
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = TRUE
PulseAudio_FOUND = 1
PATH = /home/dltc/workspace/webrtc-streamer/../webrtc/src/third_party/llvm-build/Release+Asserts/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
-- Found Git: /usr/bin/git (found version "2.34.1") 
fatal: cannot change to 'civetweb': No such file or directory
fatal: cannot change to 'live555helper': No such file or directory
VERSION = v0.8.8-dirty/Linux-x86_64 civetweb@ webrtc@b00c469cad-dirty live555helper@
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
RTMP_FOUND =  RTMP_INCLUDE_DIRS= RTMP_LIBRARY_DIRS= RTMP_LIBRARIES=
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
x live/
x live/liveMedia/
x live/groupsock/
x live/UsageEnvironment/
x live/BasicUsageEnvironment/
x live/testProgs/
x live/mediaServer/
x live/proxyServer/
x live/hlsProxy/
x live/WindowsAudioInputDevice/
x live/configure
x live/config.armeb-uclibc
[... 1480 lines cut ...]
/usr/bin/ld: main.cpp:(.text.startup+0xb95): undefined reference to `Json::Value::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: main.cpp:(.text.startup+0xba4): undefined reference to `Json::Value::operator[](char const*)'
/usr/bin/ld: main.cpp:(.text.startup+0xbaf): undefined reference to `Json::Value::operator=(Json::Value&&)'
/usr/bin/ld: main.cpp:(.text.startup+0xbb7): undefined reference to `Json::Value::~Value()'
/usr/bin/ld: main.cpp:(.text.startup+0xc4d): undefined reference to `Json::operator<<(std::ostream&, Json::Value const&)'
/usr/bin/ld: main.cpp:(.text.startup+0xdbf): undefined reference to `Json::Value::operator[](char const*)'
/usr/bin/ld: main.cpp:(.text.startup+0xf20): undefined reference to `Json::Value::~Value()'
/usr/bin/ld: main.cpp:(.text.startup+0x1689): undefined reference to `cricket::StunServer::StunServer(rtc::AsyncUDPSocket*)'
/usr/bin/ld: main.cpp:(.text.startup+0x1848): undefined reference to `cricket::TurnServer::TurnServer(webrtc::TaskQueueBase*)'
/usr/bin/ld: main.cpp:(.text.startup+0x1938): undefined reference to `cricket::TurnServer::AddInternalServerSocket(rtc::Socket*, cricket::ProtocolType, std::unique_ptr<rtc::SSLAdapterFactory, std::default_delete<rtc::SSLAdapterFactory> >)'
/usr/bin/ld: main.cpp:(.text.startup+0x19eb): undefined reference to `cricket::TurnServer::AddInternalSocket(rtc::AsyncPacketSocket*, cricket::ProtocolType)'
/usr/bin/ld: main.cpp:(.text.startup+0x1b3e): undefined reference to `cricket::TurnServer::SetExternalSocketFactory(rtc::PacketSocketFactory*, rtc::SocketAddress const&)'
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/rtpvideocapturer.cpp.o: in function `LiveVideoSource<SDPClient>::onH265Data(unsigned char*, long, long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
rtpvideocapturer.cpp:(.text._ZN15LiveVideoSourceI9SDPClientE10onH265DataEPhllRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN15LiveVideoSourceI9SDPClientE10onH265DataEPhllRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x4c): undefined reference to `webrtc::H265::FindNaluIndices(rtc::ArrayView<unsigned char const, -4711l>)'
[... cut ...]
/usr/bin/ld: rtspvideocapturer.cpp:(.text._ZN15LiveVideoSourceI14RTSPConnectionE10onH265DataEPhllRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN15LiveVideoSourceI14RTSPConnectionE10onH265DataEPhllRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x5b4): undefined reference to `webrtc::H265SpsParser::ParseSps(rtc::ArrayView<unsigned char const, -4711l>)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:246: webrtc-streamer] Error 1
make[1]: *** [CMakeFiles/Makefile2:124: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
@mpromonet

CalebShiffe, 2025-01-04:
@mpromonet

closed 2025-01-18 by mpromonet