mpromonet/webrtc-streamer #762: No ARMv6 docker manifest available?

jds11111, 2025-02-18:
**Describe the bug**
Spun up a RPi Zero W, installed the 32bit OS and docker. Checked that docker works with 'hello world'
I am probably missing something stupid, but using the recommended docker command, it says that there is no ARMv6 docker image:
```
$ docker run -p 8000:8000 -it mpromonet/webrtc-streamer
Unable to find image 'mpromonet/webrtc-streamer:latest' locally
latest: Pulling from mpromonet/webrtc-streamer
docker: no matching manifest for linux/arm/v6 in the manifest list entries.
See 'docker run --help'.
```
According to what I found here, it should exist.

**To Reproduce**
Steps to reproduce the behavior:
1. Start fresh RPi0W with latest RPiOS 32bit
2. Install docker, and check that it works
3. Try to spin up the docker container
4. Get the error message above that the manifest is missing

**Desktop (please complete the following information):**
 - OS: RPiOS Lite Bookworm 32bit
 - Browser: None
 - Docker version 27.5.1, build 9f9e405

**Additional context**
Brand new headless OS install, fully updated, nothing else installed

mpromonet, 2025-02-19:
Hi @jds11111 

It seems manifest point on something wrong, you can try to use mpromonet/webrtc-streamer:latest-armv6, or using gcr.io registry 

Best Regards
Michel

jds11111, 2025-02-24:
Thanks! That makes it work.

closed 2025-02-24 by jds11111