mpromonet/webrtc-streamer #769: can h265 video supported now?

SpringSan, 2025-03-14:
can h265 video supported now? If it supported and how to use?

mpromonet, 2025-03-14:
Hi @SpringSan 
It might works with hardware supporting h265 decoding, with -o option to forward encoding frames.
Best regards
Michel

SpringSan, 2025-03-17:
hey.The common chrome browser does not have the codec. So anyone if want to play h265 videos can download chromium from [,](https://github.com/StaZhu/enable-chromium-hevc-hardware-decoding/releases) which has the codec.

lipengsen0812, 2025-05-20:
@SpringSan Hello, do you know the specific version of the Chromium browser that supports H265

mpromonet, 2026-10-13:
Hi,
I am closing this as it is working, latest with -o option on chrome 155 is displaying h265,
Best Regards,
Michel.

closed 2026-10-13 by mpromonet