mpromonet/webrtc-streamer #775: Official Sample Code Doesn't Work and my Minimal Working Setup

fullmooooon, 2025-04-07:
[Bad Sample Code 1](https://github.com/mpromonet/webrtc-streamer#:~:text=A%20short%20sample%20HTML%20page%20using%20webrtc%2Dstreamer%20running%20locally%20on%20port%208000%3A)
I've spent hours trying to make this sample work, but unfortunately, I can't get the video to play at all. It's possible I'm missing something, but I believe this example might be outdated or incomplete.

[image]

[Bad Sample Code 2](https://hub.docker.com/r/mpromonet/webrtc-streamer#:~:text=html%3E%0A%3Chead%3E%0A%3Cscript%20src%3D%22-,libs/request.min.js,-%22%20%3E%3C/script%3E%0A%3Cscript%20src%3D%22libs)
I couldn’t find any information about what libs/request.min.js is or where to get it. The link points to a non-existent or undocumented script.

[image]

### My Minimal Working Example
In case it helps others who are also struggling to get a basic example running, here’s the minimal setup that finally worked for me.

#### Docker Command
```
docker run -p 8000:8000 -it -v "C:/config.json:/app/config.json" mpromonet/webrtc-streamer -vvvv -C /app/config.json
```
#### config.json
```json
{
    "urls":{       
        "Camera1" : { "video": "rtsp://username:password@192.168.0.1:554/cam/realmonitor?channel=1&subtype=0"}
    }
}

```
#### HTML Frontend
```
<html>
<head>
    <title>WebRTC Streamer using WHEP</title>
    <script src="./libs/whep-video.component.js"></script>
</head>
<body>
  <div id="content">
  </div>  
  <script>
        let streamElement = document.getElementById("content");
        streamElement.removeChild(streamElement.firstChild);
        let whep = document.createElement('whep-video');
        whep.setAttribute('muted', true);
        whep.setAttribute('autoplay', true);
        whep.setAttribute('src',
        `http://localhost:8000/api/whep?url=Camera1&audio=undefined&options=rtptransport%3dtcp%26timeout%3d60`
        );
        streamElement.appendChild(whep);
  </script>
</body>
</html>

```

I hope to help anyone like me who can't even work with samples

mpromonet, 2025-04-07:
Hi @fullmooooon,

I updated dockerhub readme that was out of date.

Best Regards,
Michel.

fullmooooon, 2025-04-08:
I've just figured out a few more things that might help others running into similar issues.

It turns out the official sample code only works correctly when `network_mode: host` is enabled. 

### docker-compose.yaml fragment
```
webrtc-streamer:
  image: mpromonet/webrtc-streamer:latest
  container_name: webrtc-streamer
  volumes:
    - ./webrtc-config/config.json:/app/config.json:ro
  command: -C /app/config.json -H 0.0.0.0:8082
  network_mode: host
  restart: unless-stopped
```
Without `network_mode: host`, the container seems unable to access the RTSP stream, likely due to network isolation in Docker's default bridge mode.

**Interestingly**, when I use my own custom HTML page with `whep-video.component.js`, the video stream plays fine **regardless of whether I'm using** `host` **or bridge mode** (at the same computer). This suggests that the WHEP-based player is more robust or handles networking differently.

Hope this helps others avoid hours of trial and error. It would be great if the official samples could include a note about the importance of network_mode: host for RTSP access.

Thanks again for this project!

closed 2025-04-08 by fullmooooon