mpromonet/webrtc-streamer #778: can not play webrtc stream

dannyxu2015, 2025-04-18:
Used ffmpeg to generate local rtsp stream, and vlc play well which url is `rtsp://localhost:8554/live`,

 then start the webrtc-streamer by docker  
`docker run --name webrtc-streamer -d -p 8000:8000  mpromonet/webrtc-streamer -n steamer -u rtsp://localhost:8554/live`

access 'http://localhost:8000/webrtcstreamer.html?video=streamer&options=rtptransport%3Dtcp%26timeout%3D600&' with chrome, can not play the video

the docker log:

[image]

the console capture:

[image]

Looking forward your help, Thanks!

mpromonet, 2025-04-18:
Hi @dannyxu2015 

localhost is the docker itself, not the host that run the docker.

Best Regards 
Michel

dannyxu2015, 2025-04-21:
Thanks for the quick reply, '--net host' seems not work on Mac, so I use the 'host.docker.internal' for access host network

`docker run --name webrtc-streamer -d -p 8000:8000 mpromonet/webrtc-streamer -n xxx -u rtsp://host.docker.internal:8554/live`

But still can't access the video, log:
```
Version:v0.8.9-21-gf9b577c/Linux-x86_64 civetweb@v1.16 webrtc@4030180fc5-dirty live555helper@bd5f554
{
	"urls" :
	{
		"xxx" :
		{
			"video" : "rtsp://host.docker.internal:8554/live"
		}
	}
}Logger level:4
HTTP Listen at 0.0.0.0:8000
Requested URL : rtsp://host.docker.internal:8554/live
Start playing sink for "video/H264" subsession
Close session: video/H264
Requested URL : rtsp://host.docker.internal:8554/live
Start playing sink for "video/H264" subsession
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
[h264 @ 0x7f3884002f00] non-existing PPS 0 referenced
[h264 @ 0x7f3884002f00] decode_slice_header error
[h264 @ 0x7f3884002f00] no frame!
Close session: video/H264
```

mpromonet, 2025-05-04:
Hi @dannyxu2015 

From the errors, it seems h264 stream header cannot be decoded, it may be a problem in your input stream or in h264 decoder, or in the code between.

Best regards 
Michel