mpromonet/webrtc-streamer #779: The earpiece is grayed out and cannot turn on the sound

Gciky, 2025-04-23:
I use VLC player to play RTSP stream normally and can hear sound.  The following image shows the video information displayed by the codec :
[image]
but with the latest version of webrtc-streamer, I can only play video stream and the bottom earpiece is grayed out, so I cannot hear sound
[image]

Looking forward your help, Thanks!

mpromonet, 2025-04-24:
Hi @Gciky 

The audio codec might be not supported or not implemented.

Best regards 
Michel

Gciky, 2025-04-25:
Thanks for the quick reply! I can control the earpiece when playing the same RTSP video stream using version 0.8.0, so I would like to ask if version 0.8.10 is not supported.

mpromonet, 2025-04-25:
Hi,
I guess you need to set the audiourl parameter.
Best regards 
Michel

Gciky, 2025-04-26:
> Hi, I guess you need to set the audiourl parameter. Best regards Michel

Thanks, I will go through the documentation to find out how to configure parameters. 
Best regards
Gciky

closed 2025-05-07 by mpromonet