mpromonet/webrtc-streamer #780: Screen goes crazy after adding -o

flowersBloom826, 2025-04-25:
I get a screen with a pattern after adding the -o attribute, and even after adjusting various parameters, it still doesn't work. Is there any good solution

[image]

mpromonet, 2025-04-25:
Hi @flowersBloom826 
I think you need to investigate.
Best regards 
Michel

flowersBloom826, 2025-04-25:
> Hi [@flowersBloom826](https://github.com/flowersBloom826) I think you need to investigate. Best regards Michel

No idea, do you have any good ideas?

flowersBloom826, 2025-04-27:
> https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h#L107Check the buffer in the middle, and see how to handle SPS/PPS and buffer sendinghttps://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h#L157C21-L157C28 ，What does this mean, and how can it be specifically resolved, boss?

Do I need to manually modify the source code and then pack it?

flowersBloom826, 2025-04-27:
The issue with the camera's backend configuration provided by the Huawei manufacturer has been resolved. Thank you.

closed 2025-04-27 by flowersBloom826

oiDoraemeng, 2025-11-27:
[image]

I encountered the same problem as you. When I run webrtc-streamer.exe -o like this, the screen turns green. When I run webrtc-streamer.exe directly, the screen doesn't turn green but it will flicker and lag

[image]