mpromonet/webrtc-streamer #782: Optimizing webrtc-streamer

glisa-techno, 2025-04-29:
Hi @mpromonet,

I am new to WebRTC and am currently using the WebRTC-Streamer to stream my RTSP feed. My goal is to stream only the RTSP video to a webpage, as I am running this on an embedded system with limited 128 MB RAM.

To achieve this, I want to focus exclusively on the RTSP to WebRTC functionality, minimizing the size of the binary and reducing CPU usage. I’ve tried disabling unnecessary features, but it hasn’t produced the desired results.

Could you provide the compilation steps to disable other features, or guide me on how to modify the source code?

Has anyone implemented just the RTSP to WebRTC functionality in WebRTC-Streamer? Any guidance would be greatly appreciated.

NolanMaverick, 2025-05-02:
Hello @mpromonet ,
                             I'm also working on the same, I could able to compile and stream in Ubuntu x86 system, but can I know how to cross compile for arm?

mpromonet, 2025-05-04:
Hi @NolanMaverick 

To cross compile for arm64 you can get steps from dockerfile https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile.arm64

Best regards 
Michel