mpromonet/webrtc-streamer #784: Issue with cross compiling for arm

glisa-techno, 2025-05-05:
Hi @mpromonet, I am trying to cross-compile the WebRTC Streamer for the ARM architecture. With CMake, I am not getting any errors, but when compiling using the make command, I am encountering the following error: "No such file."
 `webrtc-streamer/../webrtc/src/third_party/boringssl/src/include/openssl/span.h:27:23: fatal error: string_view: No such file or directory
 #include <string_view>
                       ^
compilation terminated.
make[2]: *** [live555helper/CMakeFiles/liblive555helper.dir/build.make:271: live555helper/CMakeFiles/liblive555helper.dir/__/live/liveMedia/AC3AudioFileServerMediaSubsession.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:263: live555helper/CMakeFiles/liblive555helper.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
`

But the file **string_view** is present in the WebRTC SDK. Can help me to solve this issue while cross compiling?
 I have followed the steps in this file [(https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile.arm64)] to cross-compile for ARM.
`cmake -DCMAKE_SYSTEM_PROCESSOR=arm -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_C_COMPILER=arm-ca9-linux-gnueabihf-gcc -DCMAKE_CXX_COMPILER=arm-ca9-linux-gnueabihf-g++ -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=NEVER -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_PACKAGE=ONLY -DWEBRTCDESKTOPCAPTURE=OFF . && make`

mpromonet, 2025-05-05:
Hi @glisa-techno 

It seems your hardware is more armv7 than arm64. If you look to https://github.com/mpromonet/webrtc-streamer/blob/master/CMakeLists.txt#L20, you should see, depending on CMAKE_SYSTEM_PROCESSOR different choice are done (with WebRTC sysroot or not, and other things.....).
The value "arm" you set is not managed and goes to default, you might start with "armv7l", and fix the problems you will meet.

Best Regards,
Michel.

glisa-techno, 2025-05-08:
Hi @mpromonet,

Thank you for your response.
I am now able to compile webrtc-streamer with armv7, but the compiled binary is not working on my embedded board. After several attempts and some debugging, I discovered that the binary is being compiled using the cross-compiler version 9.5.0, whereas the board's cross-compiler version is 6.5.0.

Could you help me compile the SDK and webrtc-streamer with the 6.5.0 version of the cross-compiler? I attempted to modify the SDK source files to compile with the 6.5.0 cross-compiler, but I am looking for a more generic way to compile the source code so that it works on my board.

Is there something I might be missing that could be causing the binary to fail?

glisa-techno, 2025-05-10:
> Hi [@mpromonet](https://github.com/mpromonet),
>
> Thank you for your response. I am now able to compile webrtc-streamer with armv7, but the compiled binary is not working on my embedded board. After [...]

or any patch I have to apply to cross compile with lower version.

mpromonet, 2025-05-10:
Hi @glisa-techno,
You may look or ask to https://issues.webrtc.org/.
It is possible that if you are using an old compiler, many changes may be needed.
You can also compile with clang...
Best Regards,
Michel.