mpromonet/webrtc-streamer #794: When using playback monitoring equipment, it will get stuck every two or three seconds, and there is a solution, using a local area network

hansongLove, 2025-07-21:
When using playback monitoring equipment, it will get stuck every two or three seconds, and there is a solution, using a local area network。help，me。thanks

SnowDragonLiang, 2025-07-23:
You can try using version V0.8.0, I didn't encounter this issue after replacing it

SnowDragonLiang, 2025-07-23:
> 使用播放监控设备时，每隔两三秒就会卡一次，有解决办法，使用局域网。帮帮我吧。谢谢
You can try using version V0.8.0, I didn't encounter this issue after replacing it

SnowDragonLiang, 2025-07-23:
> When using playback monitoring equipment, it will get stuck every two or three seconds, and there is a solution, using a local area network。help，me。thanks
You can try using version V0.8.0, I didn't encounter this issue after replacing it

hansongLove, 2025-07-31:
> > 使用回放监控设备时，每隔两三秒就会卡一次，有解决办法，用局域网。帮帮我吧。谢谢
> > 您可以尝试使用V0.8.0版本，我更换了之后就没再遇到这个问题了

Thank you for your reply, but I still get stuck using the v0.8.0 version, and the version I need to be compatible with win7, which seems not ideal

closed 2026-10-31 by mpromonet